polarfire-soc/polarfire-soc-documentation #1: Debugging bare metal on mpfs-icicle-kit-es with GDB

Fabien-Chouteau, 2020-07-17:
Hello,

Is there a way to get a GDB remote connection to the onboard FlashPro? For instance with openocd.

Regards,

Cyril-Jean, 2020-07-17:
If you are one of the lucky people with a pre-production Icicle Kit, you need to use an external FlashPro JTAG probe. Then you can use openOCD.
You will be able to use the embedded FlashPro (USB connector J33) with the production boards coming at the end of the summer.

Fabien-Chouteau, 2020-07-17:
Thank you @Cyril-Jean,

Where can I find an openOCD release and scripts that supports FlashPro and the mpfs-icicle-kit-es?

Cyril-Jean, 2020-07-17:
Easiest is to the use SoftConsole v6.3. You can download the installer from the following links:
Linux host:http://soc.microsemi.com/download/reg/download.aspx?p=f%3dSoftConsole_v6_3_LIN
Windows host: http://soc.microsemi.com/download/reg/download.aspx?p=f%3dSoftConsole_v6_3_WIN

Fabien-Chouteau, 2020-07-21:
It works, thank you @Cyril-Jean 

For the record I used this command line:
```shell
openocd/bin/openocd --command "set DEVICE MPFS" --file openocd/share/openocd/scripts/board/microsemi-riscv.cfg
```

closed 2020-07-21 by Fabien-Chouteau