polarfire-soc/polarfire-soc-documentation #2: Networking

bloudraak, 2020-10-26:
I received my two PolarFire SoC Icicle Kits last week and followed the quick start to verify it works as expected. I then connected one using an ethernet cable to a switch and verified I could SSH into the host. When I connected the second board, SSH no longer worked, giving me an error:

```
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
@    WARNING: REMOTE HOST IDENTIFICATION HAS CHANGED!     @
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
IT IS POSSIBLE THAT SOMEONE IS DOING SOMETHING NASTY!
Someone could be eavesdropping on you right now (man-in-the-middle attack)!
It is also possible that a host key has just been changed.
The fingerprint for the RSA key sent by the remote host is
SHA256:Fx********n8.
Please contact your system administrator.
Add correct host key in /Users/wstrydom/.ssh/known_hosts to get rid of this message.
Offending RSA key in /Users/wstrydom/.ssh/known_hosts:1
RSA host key for 192.168.100.5 has changed and you have requested strict checking.
Host key verification failed.
```

Upon further investigation, it seems both boards report the same MAC address. As for documentation, I don't see any documentation to

* If there is anything I have to do to have each board have its own MAC address.
* Where log files are located of things that happen during boot. 
* How to remotely update the Linux kernel and supporting software. 

My first objective is to have the two boards run Linux and that I can manage them remotely (over SSH) and compile code for RISC-V (just "hello world" in C for now). The purpose of two boards was so I can compile on one, and deploy on the other.  But it seems I'm missing something given that both boards are using the same MAC address. Hoping it's just a gap in the documentation, or that it's well hidden and I just failed to find it.

Cyril-Jean, 2020-10-27:
First, can you please try out the programming files and images supplied in this document?: https://github.com/polarfire-soc/polarfire-soc-documentation/blob/master/boards/mpfs-icicle-kit-es/updating-icicle-kit/updating-icicle-kit-design-and-linux.md

The MAC address is modified by U-Boot based on the PolarFire SoC device's serial number.

Cyril-Jean, 2021-01-26:
@bloudraak, Can you confirm that the updated image fix the issue?

closed 2022-04-28 by Cyril-Jean