polarfire-soc/polarfire-soc-documentation #3: link for FlashPro files and Linux images corrupted

OrlyShdema, 2020-12-08:
Hi ,

links for FlashPro files and Linux images  are corrupted and the files can not be downloaded..
Pls. advise.

Thanks,
Orly

OrlyShdema, 2020-12-08:
for the icicle kit

hughbreslin, 2020-12-08:
Hey Orly, 

thanks for letting us know, we'll try and get those links fixed asap 🙂 

Hugh

hughbreslin, 2020-12-08:
Hi Orly, 

could you try using a different browser and let me know if the links work for you? I primarily use Chrome and that won't open the links but trying to download the files using Internet Explorer does.

kevinscully100, 2020-12-08:
The files download and unzip ok for me using Firefox

OrlyShdema, 2020-12-08:
with windows explorer it works
with chrome it doesn't work.

hughbreslin, 2021-01-12:
Hey @OrlyShdema we're providing these files as assets with the [releases](https://github.com/polarfire-soc/icicle-kit-reference-design/releases) now, can you check this is working for you? 🙂

Cyril-Jean, 2021-01-26:
Resolved by providing FPGA programming files and Linux images as GitHub assets.

closed 2021-01-26 by Cyril-Jean