polarfire-soc/polarfire-soc-documentation #4: SDCard boot, detected but still booting from eMMC

josemonsalve2, 2020-12-15:
Hello, 

I am having some issues booting from the SD Card. HSS seems to be recognizing the SD card, but the HSS keeps initializing the eMMC linux. I have the latest Linux Image and HSS [from here](https://github.com/polarfire-soc/polarfire-soc-documentation/blob/master/boards/mpfs-icicle-kit-es/updating-icicle-kit/updating-icicle-kit-design-and-linux.md).

The following is the HSS output

```
[1.875738] HSS_E51_Banner(): PolarFire(R) SoC Hart Software Services (HSS) - version 0.99.14
(c) Copyright 2017-2020 Microchip Corporation.

[1.890320] HSS_E51_Banner(): incorporating OpenSBI - version 0.6
(c) Copyright 2019-2020 Western Digital Corporation.

[1.902993] HSS_PrintBuildId(): Build ID: 6f420443dad00da179fbfe3774ad3ab6b5da8cca
[1.912107] HSS_PrintToolVersions(): Built with the following tools: 
 - riscv64-unknown-elf-gcc (xPack GNU RISC-V Embedded GCC (Microsemi SoftConsole build), 64-bit) 8.3.0
 - GNU ld (xPack GNU RISC-V Embedded GCC (Microsemi SoftConsole build), 64-bit) 2.32

[1.937018] HSS_MemTestDDRFast(): DDR size is 1 GiB
[1.943259] HSS_MMCInit(): Attempting to select SDCARD... Passed
Press a key to enter CLI, ESC to skip
Timeout in 5 seconds
.....
7.204711] HSS_TinyCLI_Parser(): CLI check timeout
[7.210873] IPI_QueuesInit(): Initializing IPI Queues (9000 bytes @ 8000e40)...
[7.219467] HSS_PMP_Init(): Initializing PMPs
[7.225109] HSS_BootInit(): Initializing Boot Image..
[7.231445] getBootImageFromMMC_(): Preparing to copy from MMC to DDR ...
[7.239517] getBootImageFromMMC_(): Attempting to read image header (1552 bytes) ...
[7.248796] GPT_ValidateHeader(): Validated GPT Header ...
[7.289688] GPT_ValidatePartitionEntries(): Validated GPT Partition Entries ...
[7.299817] copyBootImageToDDR_(): Copying 456728 bytes to 0xB0000000
[11.64473] copyBootImageToDDR_(): Calculated CRC32 of image in DDR is c23d8708
[11.157999] HSS_BootInit():  boot image passed CRC
[11.164162] HSS_BootInit(): Boot image set name: "PolarFire-SoC-HSS::U-Boot"
[11.172582] HSS_BootInit(): Boot Image registered...
[11.178918] HSS_Boot_RestartCore(): called for all harts
[11.185602] RunStateMachine(): boot_service(u54_1)::Init -> boot_service(u54_1)::SetupPMP
[11.195410] RunStateMachine(): boot_service(u54_2)::Init -> boot_service(u54_2)::SetupPMP
[11.205218] RunStateMachine(): boot_service(u54_3)::Init -> boot_service(u54_3)::SetupPMP
[11.215027] RunStateMachine(): boot_service(u54_4)::Init -> boot_service(u54_4)::SetupPMP
[11.224835] RunStateMachine(): usbdmsc_service::init -> usbdmsc_service::idle
[11.233602] RunStateMachine(): boot_service(u54_1)::SetupPMP -> boot_service(u54_1)::SetupPMPComplete
[11.244452] RunStateMachine(): boot_service(u54_2)::SetupPMP -> boot_service(u54_2)::SetupPMPComplete
[11.255302] RunStateMachine(): boot_service(u54_3)::SetupPMP -> boot_service(u54_3)::SetupPMPComplete
[11.266153] RunStateMachine(): boot_service(u54_4)::SetupPMP -> boot_service(u54_4)::SetupPMPComplete
[11.277002] RunStateMachine(): boot_service(u54_1)::SetupPMPComplete -> boot_service(u54_1)::ZeroInit
[11.287852] RunStateMachine(): boot_service(u54_2)::SetupPMPComplete -> boot_service(u54_2)::ZeroInit
[11.298702] RunStateMachine(): boot_service(u54_3)::SetupPMPComplete -> boot_service(u54_3)::ZeroInit
[11.309552] RunStateMachine(): boot_service(u54_4)::SetupPMPComplete -> boot_service(u54_4)::ZeroInit
[11.320402] RunStateMachine(): boot_service(u54_1)::ZeroInit -> boot_service(u54_1)::Download
[11.330558] RunStateMachine(): boot_service(u54_2)::ZeroInit -> boot_service(u54_2)::Download
[11.340713] RunStateMachine(): boot_service(u54_3)::ZeroInit -> boot_service(u54_3)::Download
[11.350869] RunStateMachine(): boot_service(u54_4)::ZeroInit -> boot_service(u54_4)::Download
[11.361024] RunStateMachine(): boot_service(u54_2)::Download -> boot_service(u54_2)::Idle
[11.370833] RunStateMachine(): boot_service(u54_3)::Download -> boot_service(u54_3)::Idle
[11.380641] RunStateMachine(): boot_service(u54_4)::Download -> boot_service(u54_4)::Idle
[11.415656] RunStateMachine(): boot_service(u54_1)::Download -> boot_service(u54_1)::Wait
[11.425465] boot_download_chunks_onExit(): boot_service(u54_1)::u54_2:sbi_init 80200000
[11.434840] boot_download_chunks_onExit(): boot_service(u54_1)::u54_3:sbi_init 80200000
[11.444215] boot_download_chunks_onExit(): boot_service(u54_1)::u54_4:sbi_init 80200000
[11.453589] boot_download_chunks_onExit(): boot_service(u54_1)::u54_1:sbi_init 80200000
[11.462963] boot_wait_onEntry(): boot_service(u54_1)::Checking for IPI ACKs: - -
[11.471730] boot_wait_handler(): boot_service(u54_1)::Checking for IPI ACKs: ACK/IDLE ACK
[11.481278] RunStateMachine(): boot_service(u54_1)::Wait -> boot_service(u54_1)::Idle

U-Boot 2020.07 (Nov 27 2020 - 10:24:20 +0000)

CPU:   rv64imafdc
Model: Microchip PolarFire-SoC
DRAM:  1 GiB
MMC:   sd@20008000: 0
In:    serial@20000000
Out:   serial@20000000
Err:   serial@20000000
Net:
Warning: ethernet@20112000 using MAC address from DT
eth0: ethernet@20112000
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr.uimg
1104 bytes read in 13 ms (82 KiB/s)
## Executing script at 88100000
4043924 bytes read in 349 ms (11.1 MiB/s)
## Copying 'fdt@microchip_icicle-kit-es-a000-microchip.dtb' subimage from FIT image at 88300000 ...
sha256+    Loading part 253 ... OK
## Loading kernel from FIT Image at 88300000 ...
   Using 'conf@microchip_icicle-kit-es-a000-microchip.dtb' configuration
   Trying 'kernel@1' kernel subimage
     Description:  Linux kernel
     Type:         Kernel Image
     Compression:  gzip compressed
     Data Start:   0x88300100
     Data Size:    4033939 Bytes = 3.8 MiB
     Architecture: RISC-V
     OS:           Linux
     Load Address: 0x80200000
     Entry Point:  0x80200000
     Hash algo:    sha256
     Hash value:   15d16c0ea770fdde5f1f5084cf419f521d2c4eb9382da23eccfca01c8e295293
   Verifying Hash Integrity ... sha256+ OK
## Flattened Device Tree blob at 88000000
   Booting using the fdt blob at 0x88000000
   Uncompressing Kernel Image
   Using Device Tree in place at 0000000088000000, end 0000000088004f07

Starting kernel ...

```

As you can see, the attempt for **SDCARD passes**. However, when I check the other tty for what linux it is being loaded, it is still the linux that is originally in the eMMC. 

It's good to mention that I have updated both the SD card and the eMMC linux to the latest version. 

I am not sure what else to check. I tried to go over the HSS code but it is beyond my current knowledge. 

Thanks in advance.

josemonsalve2, 2020-12-15:
It's probably worth mention that I have tried two SD cards by now:

1. 8GB SDHC Card SanDisk class 4
2. 128 GB SDXC Card SanDisk Class 10

hughbreslin, 2020-12-15:
Hey @josemonsalve2 - we had a look into this and heres what we can see:

It looks like this occurs during Linux boot – what I can see on my end is the image from SD card being loaded and when the kernel comes up and starts loading drivers it brings up the GPIO driver. Part of the GPIO driver starting up is that it initializes the GPIOs to 0 – unfortunately, this looks like the culprit as its setting GPIO2_0 to 0 and re-selecting the eMMC.

To resolve this without updating the driver or trying to build in a solution for the GPIO we’ve changed the reference design to use a register. This means that you need to write either 0 or 1 to address 0x4F00_0000 to select eMMC or SD card which will not be reset by Linux. The reference design update will be pushed out in a few days and I’ll respond here when it’s available and we’ll also be pushing out an update for the HSS to accommodate this change. They’ll be available in the 2020.12 tag.

I’m sorry for the confusion and disruption this might have caused.

josemonsalve2, 2020-12-16:
Ah! I’m si glad it’s not my fault and I was able to catch a real bug. Let me know when the PR happens. I want to see what the changes look like.  

Happy holidays!

hughbreslin, 2020-12-16:
The [tag](https://github.com/polarfire-soc/icicle-kit-reference-design/releases/tag/2020.12) is live now and we have an asset with the FlashPro express job.

Check out [this commit](https://github.com/polarfire-soc/icicle-kit-reference-design/commit/851e35845bf052e012d17aeb87a107cf10f74857) to see the changes 🙂

josemonsalve2, 2020-12-16:
Looking good over here. I was able to boot the 128 GB SD card. 

I was able to boot with the HSS 0.99.15 by running the FPexpress job in the 2020.12 tag. I am still using the same linux image I had before, so no change on the Linux side. 

Is running the FPexpress job equivalent to re-building the HSS from source? Or should I also do that? 

Thanks

Jose

josemonsalve2, 2020-12-16:
I don't think I should post this question here, but I'm gonna use the inertia of this post. Should I consider using a heat sink on my ICICLE?

hughbreslin, 2020-12-17:
Woo thats great, glad the issues resolved 👍 

When we create the FPExpress job we also built the HSS separately and included it in the job file as a client and used the source code from the HSS 2020.12 tag - so you don't need to rebuild the HSS. Whenever you run the job you'll program the eNVM with whatever version of the HSS is included in the job.

In terms of the heat sink question, you're the first one to ask as far as I know so my question is why do you think you need one? If you can email me some details I can get you some more answers internally - hugh.breslin@microchip.com

josemonsalve2, 2021-01-13:
Thanks for all the help!

closed 2021-01-13 by josemonsalve2