polarfire-soc/polarfire-soc-documentation #5: About Win32DiskImager in your docs

bztsrc, 2021-01-18:
Hi,

I've seen that you recommend a rather old piece of software all over your documentation. May I suggest another one instead?
https://bztsrc.gitlab.io/usbimager/

It has several advantages over Win32DiskImager:
- portable executable, no installation required (no dependencies)
- it can directly read .wic.gz files (no need to uncompress images first)
- it can write to SD as well as to eMMC devices (and to other USB storage)
- it's available and very well tested on Windows as well as on Linux
- has an easy to use, very simple interface, translated to 17 languages
- Open Source, MIT licensed (you can use it freely)

Cheers,
bzt

Cyril-Jean, 2021-01-26:
Thanks for the suggestion. We were/are considering switching to recommending Etcher.

bztsrc, 2021-01-30:
FYI, USBImager was written as an alternative, because Etcher has some serious, still unsolved security issues (see [this](https://github.com/balena-io/etcher/issues/2977) and [this](https://github.com/balena-io/etcher/issues/2890) and [this](https://github.com/balena-io/etcher/issues/2057) and [this](https://github.com/balena-io/etcher/issues/1718) for example), and it cannot be used in the EU as it's not GDPR compatible. They keep closing the issues, but it always popping up.

Maybe not the best idea to recommend a known spyware to your users.

Cheers,
bzt

MingyaoLiu, 2021-06-20:
Balena Etcher is open source, and the analytics code is there to be checked out freely. I can't call it a spyware as people are mad about the software trying to register with google analytics or mixpanel even with send data off. And also it seems they are sending anonymous data, not personal data, so it may not be GDPR compliant, but it doesn't sounds like a open source spyware.

hughbreslin, 2021-11-17:
Hey @bztsrc we updated our docs to point to USBImager if you want to close this issue 😄

bztsrc, 2021-11-18:
@MingyaoLiu: sending any telemetry (anonymous or not) after the user explicitly opted out, that's never good.

@hughbreslin: Thank you! I'm sure your users will be happy that you suggest a small and easy to use tool to them! Btw, you have created very nice screenshots too!

Cheers,
bzt

closed 2021-11-18 by bztsrc