polarfire-soc/polarfire-soc-documentation #7: flashing new emmc linux image

rroulleau, 2021-03-02:
I am trying to flash the new emmc image core-image-minimal-dev-icicle-kit-es-20210216165805.rootfs.wic.gz. The reboot stop in the middle when loading the linux kernel. Is there a mismatch with the new linux image and the envm image on my board as I did not re-flash the eNVM since I got the board? 
It is likely I have missed something but can not found out where...

See enclose the dump from when I flashed the emmc and log from HSS UART and U-boot&linux UART.
[issue_flashing_emmc_new_linuximage.log](https://github.com/polarfire-soc/polarfire-soc-documentation/files/6067306/issue_flashing_emmc_new_linuximage.log)

Next step I will try is to load and fresh new image generated from the rootfs-sdk env...

hughbreslin, 2021-03-02:
Hey @rroulleau I'm trying to recreate this on my end and struggling to get a failure at the same point - if this is running from an SD card you may be having an issue with the image you're using itself, there are some differences between the core-image-minimal-dev-icicle-kit-es-20210216165805.rootfs.wic.gz and core-image-minimal-dev-icicle-kit-es-sd-20210216171231.rootfs.wic.gz, the first image is for eMMC and has an eMMC bus width of 8 whereas the sd image has a bus width of 4 - which will cause an issue, but it doesn't seem to be the one you're seeing as it'll cause an issue mounting rootfs, this is where an eMMC image from an SD card crashes for me:

[    1.056065] mpfs-mailbox 37020000.mailbox: Registered MPFS mailbox controller driver
[    1.064694] pac193x 0-0010: Chip revision: 0x03
[    1.072980] pac193x 0-0010: :pac193x_prep_iio_channels: Channel 0 active
[    1.079819] pac193x 0-0010: :pac193x_prep_iio_channels: Channel 1 active
[    1.086623] pac193x 0-0010: :pac193x_prep_iio_channels: Channel 2 active
[    1.093338] pac193x 0-0010: :pac193x_prep_iio_channels: Channel 3 active
[    1.100134] pac193x 0-0010: :pac193x_prep_iio_channels: Active chip channels: 25
[    1.117658] NET: Registered protocol family 10
[    1.124367] Segment Routing with IPv6
[    1.128157] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
[    1.135112] NET: Registered protocol family 17
[    1.140322] mpfs-sys-controller soc:syscontroller: Registered MPFS system controller driver
[    1.149205] mpfs-serial-number soc:serialnum: Successfully registered mpfs serial number driver
[    1.158474] mpfs-digest soc:fpgadigest: Successfully registered mpfs fpga digest driver
[    1.168099] mpfs-rng soc:hwrandom: Successfully registered HW RNG
[    1.174465] hctosys: unable to open rtc device (rtc0)
[    1.180941] Waiting for root device /dev/mmcblk0p3...

In any case I'd flash the sd image to your card and see if it makes any difference. The other potential cause is USB, have you got anything connected up on the USB and if so can you try disconnecting that and seeing if the boot gets any further?

rroulleau, 2021-03-02:
Hello, I have been updating the eMMC memory image with this core-image-minimal-dev-icicle-kit-es-20210216165805.rootfs.wic.gz file. The SD card is unplugged on my board. I did not touch the eNVM at all since I get the eval board (Icicle kit Rev 1.0). 
On the USB port I only have the J11 and the J16 plugged in.

I will try to make a SD card using the core-image-minimal-dev-icicle-kit-es-sd-20210216171231.rootfs.wic.gz file as you suggested.

Aslo I will open another issue when flashing with the buildroot-sdk I am having a "Error: boot3 not defined", it may be related issue...

hughbreslin, 2021-09-16:
Hey @rroulleau is this problem resolved and can this issue be closed? 🙂

closed 2021-09-21 by rroulleau