polarfire-soc/polarfire-soc-documentation #8: Issue in programming eNVM using softconsole - Error: No programmers found

pathakraul, 2021-05-03:
Hi,
I am facing issue while programming the eNVM using the softconsole. I am using **Softconsole v6.5.0** and programming **HSS version 2020.12**.

My board is in the jumper settings which boots the board since there are no additional Jumper settings are mentioned in the Softconsole section. 

Please let me know if I overlooked something and missing any prerequisite.

These are the logs - 
```
[ Starting ]
[ Find fpgenprog tool ]
Info:    fpgenprog tool               is '/home/rpathak/Templates/tmp_tools/LibreSOC/LibreSOCv2021.1/Libero/bin64/fpgenprog'
[ Find RISC-V binutils tools ]
Info:    RISC-V binutils objcopy tool is '/home/rpathak/Templates/tmp_tools/Microchip/SoftConsole-v6.5/riscv-unknown-elf-gcc/bin/riscv64-unknown-elf-objcopy'
Info:    RISC-V binutils readelf tool is '/home/rpathak/Templates/tmp_tools/Microchip/SoftConsole-v6.5/riscv-unknown-elf-gcc/bin/riscv64-unknown-elf-readelf'
Info:    working directory is '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default'
Info:    ELF file is 'hss.elf'
Info:    get boot address and image size from 'hss.elf' for bootmode 1
Info:    /home/rpathak/Templates/tmp_tools/Microchip/SoftConsole-v6.5/riscv-unknown-elf-gcc/bin/riscv64-unknown-elf-readelf --program-headers hss.elf
Info:    boot address = 0x20220000
Info:    program size = 112064
[ Create fpgenprog project... ]
[ Configure boot mode and NVM client ]
prog_bootaddr=20220000
envm_lo=0x20220000
start_page=0
[ Generate bitstream... ]
Software Version: 2021.1.0.17
creating folder: /home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/proj_fp
Created new project '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/proj_fp/proj_fp.pro'
Project saved.
Project closed.
Software Version: 2021.1.0.17
Opened '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/proj_fp/proj_fp.pro'
PDB file '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/target.pdb' has been loaded successfully.
DESIGN : TOP;  CHECKSUM : 0000;  PDB_VERSION : 1.9
Load Programming Data(s) Finished : Mon May  3 11:51:03 2021 (Elapsed time 00:00:00)
Successfully exported PPD file for currently secured device: '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/proj_fp/target.ppd'; file programs eNVM.
BITS component bitstream digest: 25295f0d1d592a90b333e26e85149708208e9f8e8bc18f6c77bd62f8ad7a6866
Security component bitstream digest: e8838677258cebf6d827c37b4a53e78a05e000b4bf8f19514538e985a0280c99
eNVM component bitstream digest: e9cfcf8581788fa187d7ae53bead0d8e46446b360007c6eabdb39e3ee327399f
Finished: Mon May  3 11:51:06 2021 (Elapsed time 00:00:03)

creating folder: /home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/proj_fp/projectData
Project saved.
Project closed.
[ Program target device... ]
Console Mode = Software Version: 2021.1.0.17
Warning:  Unable to find file: '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/target.pdb'.
Warning: [80609481]:  Using local backup copy: '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/proj_fp/projectData/target.pdb'.
PDB file '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/proj_fp/projectData/target.pdb' has been loaded successfully.
DESIGN : TOP;  CHECKSUM : 0000;  PDB_VERSION : 1.9
Device not found
Opened '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/proj_fp/proj_fp.pro'
The 'open_project' command succeeded.
Warning:  Programming is already enabled for device 'target'.
The 'enable_device' command succeeded.
PPD file '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/proj_fp/target.ppd' has been loaded successfully.
DESIGN : TOP;  CHECKSUM : 0000;  PDB_VERSION : 1.0
The 'set_programming_file' command succeeded.
The 'set_programming_action' command succeeded.
Error:  No programmers found. Please check the programmer connection to the computer and ensure the drivers are properly installed.
Error:  Failed to run Action.
Exported log file /home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/proj_fp/proj_fp.log.
Error:  The command 'run_selected_actions' failed.
Error:  Failure when executing Tcl script. [ Line 5 ]
Error:  The Execute Script command failed.
Exported log file /home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/proj_fp/proj_fp.log.
Software Version: 2021.1.0.17
Warning: Unable to find file: '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/target.pdb'.
Warning: Using local backup copy: '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/proj_fp/projectData/target.pdb'.
PDB file '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/proj_fp/projectData/target.pdb' has been loaded successfully.
DESIGN : TOP;  CHECKSUM : 0000;  PDB_VERSION : 1.9
Opened '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/proj_fp/proj_fp.pro'
The 'open_project' command succeeded.
Warning: Programming is already enabled for device 'target'.
The 'enable_device' command succeeded.
PPD file '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/proj_fp/target.ppd' has been loaded successfully.
DESIGN : TOP;  CHECKSUM : 0000;  PDB_VERSION : 1.0
The 'set_programming_file' command succeeded.
The 'set_programming_action' command succeeded.
Error: No programmers found. Please check the programmer connection to the computer and ensure the drivers are properly installed.
Error: Failed to run Action.
Exported log file /home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/proj_fp/proj_fp.log.
Error: The command 'run_selected_actions' failed.
Error: Failure when executing Tcl script. [ Line 5 ]
Error: The Execute Script command failed.
Error: Failed to run action. Return code = 256
Error:   PROGRAM action failed, exiting...
Error:   check the log for errors and refer to /home/rpathak/Templates/tmp_tools/Microchip/SoftConsole-v6.5/extras/mpfs/readme.txt if necessary
[ Finished (failure) ]
```

nitindeshpande, 2021-05-03:
Hi Rahul,
 It seems the eFP6 or the external programmer is not getting detected.
 Are you using eFP6? How are the J9 and J24 placed on your kit? Were you able to program the eNVM before?

The 'enable_device' command succeeded.
PPD file '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/proj_fp/target.ppd' has been loaded successfully.
DESIGN : TOP;  CHECKSUM : 0000;  PDB_VERSION : 1.0
The 'set_programming_file' command succeeded.
The 'set_programming_action' command succeeded.
Error:  No programmers found. Please check the programmer connection to the computer and ensure the drivers are properly installed.
Error:  Failed to run Action.

pathakraul, 2021-05-03:
Hi Nitin,

This is the first time I am flashing. As per the documentation my understanding is that embedded programmer is inbuilt in this board and no external FlashPro is required. Kit was also not shipped with FlashPro.

Both J9 and J24 are closed. And PC is connected with board via J33.

Thanks
Rahul

nitindeshpande, 2021-05-03:
Can you please confirm the following?
1) icicle kit revision. You will find this printed near the PCIe connector.
[image]
2) OS version on your system
3)  Libero version and license info.

pathakraul, 2021-05-03:
Board Revision is 1.0 
Host OS - Ubuntu 20.04.

I flashed eMMC with **core-image-minimal-dev-icicle-kit-es-20210216165805.rootfs.wic.gz**  earlier but ran into an issue, here is the link [https://github.com/polarfire-soc/meta-polarfire-soc-yocto-bsp/issues/22](https://github.com/polarfire-soc/meta-polarfire-soc-yocto-bsp/issues/22). 

Microsemi-Libero_SoC_v2021.1 is the one I have installed.  Not sure how to fetch the License information.
The Icicle FPGA Design is the default one which came with the board.

nitindeshpande, 2021-05-03:
Thanks for the info, Rahul.
May be the drivers are not installed properly, when J33 is connected to the PC, do you see any FlashPro USB device connected if you do the dmesg?

pathakraul, 2021-05-03:
Yes, here is the log
```
[116223.721171] usb 1-3: USB disconnect, device number 31
[116233.635015] usb 1-3: new high-speed USB device number 32 using xhci_hcd
[116233.784049] usb 1-3: New USB device found, idVendor=1514, idProduct=200b, bcdDevice=ef.9b
[116233.784055] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[116233.784059] usb 1-3: Product: Embedded FP6b
[116233.784062] usb 1-3: Manufacturer: Microchip
[116233.784064] usb 1-3: SerialNumber: 0744DEF0
[116233.787098] hid-generic 0003:1514:200B.0018: hiddev4,hidraw9: USB HID v1.10 Device [Microchip Embedded FP6b] on usb-0000:00:14.0-3/input0
```

nitindeshpande, 2021-05-03:
Thats cool. No issues with the hardware.
It might help to look at section 2.10 of this document? 
https://coredocs.s3.amazonaws.com/Libero/2021_1/Tool/linux_setup_libero_soc_ug.pdf

pathakraul, 2021-05-03:
Ok, glanced through the document and I have few questions -
 
So far I installed the Libero SOC but only using the **fpgenprog** from that installation in the SoftConsole.
Since the document which you have given is for LibreSOC, does it means that the LibreSOC has to be used for flashing eNVM instead of SoftConsole?

Or

I just follow "Step 10—Configure Linux to Detect and Use Programmer Hardware" from the document and continue with SoftConsole afterwards.

nitindeshpande, 2021-05-03:
SoftConsole need the fpgenprog from the LiberoSoC for flashing eNVM. You can continue with SoftConsole.

pathakraul, 2021-05-03:
ok thanks. I will verify the linux setup as per the document and update

pathakraul, 2021-05-04:
I followed the steps from section "Configure Linux to Detect and Use FlashPro6 Programmer Hardware". 
This time it moved ahead but now its giving **Segmentation Fault** and LED5 starts blinking after that.

Complete logs -
```
[ Starting ]
[ Find fpgenprog tool ]
Info:    fpgenprog tool               is '/home/rpathak/Templates/tmp_tools/LibreSOC/LibreSOCv2021.1/Libero/bin64/fpgenprog'
[ Find RISC-V binutils tools ]
Info:    RISC-V binutils objcopy tool is '/home/rpathak/Templates/tmp_tools/Microchip/SoftConsole-v6.5/riscv-unknown-elf-gcc/bin/riscv64-unknown-elf-objcopy'
Info:    RISC-V binutils readelf tool is '/home/rpathak/Templates/tmp_tools/Microchip/SoftConsole-v6.5/riscv-unknown-elf-gcc/bin/riscv64-unknown-elf-readelf'
Info:    working directory is '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default'
Info:    ELF file is 'hss.elf'
Info:    get boot address and image size from 'hss.elf' for bootmode 1
Info:    /home/rpathak/Templates/tmp_tools/Microchip/SoftConsole-v6.5/riscv-unknown-elf-gcc/bin/riscv64-unknown-elf-readelf --program-headers hss.elf
Info:    boot address = 0x20220000
Info:    program size = 112064
[ Create fpgenprog project... ]
[ Configure boot mode and NVM client ]
prog_bootaddr=20220000
envm_lo=0x20220000
start_page=0
[ Generate bitstream... ]
Software Version: 2021.1.0.17
creating folder: /home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/proj_fp
Created new project '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/proj_fp/proj_fp.pro'
Project saved.
Project closed.
Software Version: 2021.1.0.17
Opened '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/proj_fp/proj_fp.pro'
PDB file '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/target.pdb' has been loaded successfully.
DESIGN : TOP;  CHECKSUM : 0000;  PDB_VERSION : 1.9
Load Programming Data(s) Finished : Tue May  4 07:13:08 2021 (Elapsed time 00:00:01)
Successfully exported PPD file for currently secured device: '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/proj_fp/target.ppd'; file programs eNVM.
BITS component bitstream digest: 25295f0d1d592a90b333e26e85149708208e9f8e8bc18f6c77bd62f8ad7a6866
Security component bitstream digest: e8838677258cebf6d827c37b4a53e78a05e000b4bf8f19514538e985a0280c99
eNVM component bitstream digest: e9cfcf8581788fa187d7ae53bead0d8e46446b360007c6eabdb39e3ee327399f
Finished: Tue May  4 07:13:10 2021 (Elapsed time 00:00:02)

creating folder: /home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/proj_fp/projectData
Project saved.
Project closed.
[ Program target device... ]
Console Mode = Software Version: 2021.1.0.17
Warning:  Unable to find file: '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/target.pdb'.
Warning: [80609481]:  Using local backup copy: '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/proj_fp/projectData/target.pdb'.
PDB file '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/proj_fp/projectData/target.pdb' has been loaded successfully.
DESIGN : TOP;  CHECKSUM : 0000;  PDB_VERSION : 1.9
===============================================================
Programmer: Embedded FP6 '0744DEF0' : Version = EF9B
===============================================================
Device not found
programmer '0744DEF0' : Embedded FlashPro6
Opened '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/proj_fp/proj_fp.pro'
The 'open_project' command succeeded.
Warning:  Programming is already enabled for device 'target'.
The 'enable_device' command succeeded.
PPD file '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/fpgenprog/proj_fp/target.ppd' has been loaded successfully.
DESIGN : TOP;  CHECKSUM : 0000;  PDB_VERSION : 1.0
The 'set_programming_file' command succeeded.
The 'set_programming_action' command succeeded.
programmer '0744DEF0' : Scan Chain...
programmer '0744DEF0' : Check Chain...
programmer '0744DEF0' : Scan and Check Chain PASSED.
@@ACTIVITY 1
programmer '0744DEF0' : device 'target' : Executing action PROGRAM
Performing FPGA action. Please wait...
eFP6 acceleration mode enabled with PPD file. Please wait...
@@ACTIVITY 2
Segmentation fault (core dumped)
Error: Failed to run action. Return code = 35584
Error:   PROGRAM action failed, exiting...
Error:   check the log for errors and refer to /home/rpathak/Templates/tmp_tools/Microchip/SoftConsole-v6.5/extras/mpfs/readme.txt if necessary
[ Finished (failure) ]
```

nitindeshpande, 2021-05-04:
eFP6 is detected by SoftConsole now. Will get back soon.

nitindeshpande, 2021-05-04:
Hi Rahul,
 It might be a better idea to install and use[ SoftConsole v2021.1](https://www.microsemi.com/product-directory/design-tools/4879-softconsole#downloads). After installation please check page 21 of the SoftConsole Release Notes.

pathakraul, 2021-05-04:
I seems its does not want me to flash it :). I installed the latest Softconsole. Used the latest HSS tag. Followed the release notes page 21.

J9 - Closed
J24 - Closed

```
./openocd -f board/microsemi-riscv.cfg
xPack OpenOCD (Microchip SoftConsole build), x86_64 Open On-Chip Debugger 0.10.0+dev-00859-g95a8cd9b5-dirty (2020-10-21-21:16)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
fpServer v17 waiting for incoming connections on the port 3334 with API v5
do_board_reset_init
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : Embedded FlashPro6 (revision B) found (USB_ID=1514:200b path=/dev/hidraw9)
Info : Embedded FlashPro6 (revision B) CM3 firmware version: F4.0
Info : clock speed 6000 kHz
Info : JTAG tap: fpga.tap tap/device found: 0x0f81a1cf (mfg: 0x0e7 (GateField), part: 0xf81a, ver: 0x0)
Info : JTAG tap: fpga.tap disabled
Info : JTAG tap: fpga.cpu enabled
Info : RISC-V IDCODE = 0x00000000
Error: dtmcontrol is 0. Check JTAG connectivity/board power.
Info : Listening on port 3333 for gdb connections
```

lsusb
```
Bus 001 Device 024: ID 1514:200b Actel Embedded FP6b
```
Console
```
17:17:48 INFO  - mpfsBootmodeProgrammer v3.3 started.
17:17:48 INFO  - "/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/bootmode1" is the output folder and the previous contents of this folder will be deleted.
17:17:48 INFO  - Selected boot mode "1 - non-secure boot from eNVM" and working in directory "/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default".
17:17:48 INFO  - Generating BIN file...
17:17:48 INFO  - Generating header...
17:17:48 INFO  - Generating HEX file...
17:17:48 INFO  - Preparing for bitstream generation...
17:17:48 INFO  - Generating bitstream...
17:17:56 INFO  - Programming the target...
17:18:01 INFO  - Console Mode = Software Version: 2021.1.0.17
Warning:  Unable to find file: '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/bootmode1/fpgenprogProject/target.pdb'.
Warning: [80609481]:  Using local backup copy: '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/bootmode1/fpgenprogProject/proj_fp/projectData/target.pdb'.
PDB file '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/bootmode1/fpgenprogProject/proj_fp/projectData/target.pdb' has been loaded successfully.
DESIGN : TOP;  CHECKSUM : 0000;  PDB_VERSION : 1.9
===============================================================
Programmer: Embedded FP6 '0744DEF0' : Version = EF9B
===============================================================
Device not found
programmer '0744DEF0' : Embedded FlashPro6
Opened '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/bootmode1/fpgenprogProject/proj_fp/proj_fp.pro'
The 'open_project' command succeeded.
Warning:  Programming is already enabled for device 'target'.
The 'enable_device' command succeeded.
PPD file '/home/rpathak/Project/workspace/microchip-polarfire/hart-software-services/Default/bootmode1/fpgenprogProject/proj_fp/target.ppd' has been loaded successfully.
DESIGN : TOP;  CHECKSUM : 0000;  PDB_VERSION : 1.0
The 'set_programming_file' command succeeded.
The 'set_programming_action' command succeeded.
programmer '0744DEF0' : Scan Chain...
programmer '0744DEF0' : Check Chain...
programmer '0744DEF0' : Scan and Check Chain PASSED.
@@ACTIVITY 1
programmer '0744DEF0' : device 'target' : Executing action PROGRAM
Performing FPGA action. Please wait...
eFP6 acceleration mode enabled with PPD file. Please wait...
@@ACTIVITY 2

17:18:01 ERROR - Running the fpgenprog PROGRAM action failed.
```

After this the LED5 constantly blinking

pathakraul, 2021-05-05:
FPExpress_bin is the part of the LibreSOC and this is the latest which is used. Is there any stable version which I can try?

nitindeshpande, 2021-05-05:
Sorry for the delay, The segfault is totally unexpected.
We are looking into it.

Did you use SoftConsole as root?

You are using fpgenprog that is part of LiberoSoC2021.1, there is an alternate way.
Install Programming and Debug v2021.1 https://www.microsemi.com/product-directory/dev-tools/5592-programming-and-debug#program 
Can you please see if that works.

pathakraul, 2021-05-05:
No softconsole is from non-root user.
I will check. 

I also wanted to know which FPGA Family I shall select for the Polarfire SOC. I am trying to apply for a evaluation license for LibreSOC.  I don't see Polarfire mentioned.
[image]

Because eventually I might need to upgrade the Icicle kit because of changes in memory map and support with latest polarfire software stack.

nitindeshpande, 2021-05-05:
Hi,
Please do run SoftConsole as root. 

PolarFire SoC is not supported with the evaluation license. https://www.microsemi.com/product-directory/licensing/4890-evaluation-license

pathakraul, 2021-05-05:
HI Nitin, I am referring the License Selector guide which takes me to the Free Licence page for MPFS250T-FCVG484

[image]

If this is not right then please guide me how can I or my organisation can obtain license and which one?

Since latest icicle-design-kit has the memory map changes and latest released HSS depends on it, I have to upgrade it using the LibreSOC.

nitindeshpande, 2021-05-05:
Hi Rahul,
 For LiberoSoC licensing its best to go through the support team. http://soc.microsemi.com/mycases. Team will guide you.

pathakraul, 2021-05-05:
Got it, thanks Nitin

nitindeshpande, 2021-05-06:
When you try to flash the eNVM with HSS using SoftConsole, by any chance, is your J33 USB connected to the PC via USB hub?

pathakraul, 2021-05-06:
No, its directly connected to system

pathakraul, 2021-05-13:
I reinstalled Ubuntu on my system and reinstalled Softconsole and LibreSOC, but same segmentation fault.

nitindeshpande, 2021-05-18:
Hi Rahul,
 We looked into the issue but we are not able to reproduce the segmentation fault that you see.
 This needs to be looked at by the fpgenprog tool team. 
 I suggest you file a case @  http://soc.microsemi.com/mycases.

pathakraul, 2021-05-18:
Sure, thanks.

Below Jumper settings are the one which I am using while flashing. These are correct right???

Jumper | Setting
-- | --
J15 | Open
J17 | Open
J24 | Open
J28 | Closed
J31 | Open
J34 | 1 &amp; 2
J35 | 1 &amp; 2
J43 | 1 &amp; 2
J45 | 1 &amp; 2
J46 | Closed

Jumper | Description
-- | --
J9 | Used to select between Embedded FlashPro6 and external FlashPro header.
&nbsp; | Closed: Embedded FlashPro6 connected to J33 micro USB port.
&nbsp; | Open: External FlashPro connected to J23 header.
J21 | JTAG nTRST interface pull down enable. Leave open.

**The MMC voltage level change requires jumpers 34 and 43 of the Icicle Kit to be changed from the 1 & 2 position to the 2 & 3 position.**

Which Ubuntu version you are using for the flashing and stuff - 20.04 or 18.04??

nitindeshpande, 2021-05-18:
That's correct, the doc page update has not reflected yet. 
If you are using design which uses v1.8 for emmc then jumper 34 and 43 should be 2&3, for 3.3 volts  jumper 34 and 43 should be 1&2.
Reference design v2021.04 uses eMMC @ 1.8v. Previous releases use eMMC @v3.3v

It is verified with 20.04 as well as 18.04.

nitindeshpande, 2021-08-11:
@pathakraul
If your query was answered, can this issue be closed?

closed 2021-08-15 by pathakraul