polarfire-soc/polarfire-soc-documentation #10: OpenOCD failed with code (-1073741819)

kowalski100, 2021-09-02:
I am trying to debug a bare-metal code via Softconsole latest version and 6.5. The debugger was working fine till yesterday morning when suddenly I started getting the following issue.

[image]

Any clue what's wrong with the openocd?

nitindeshpande, 2021-09-03:
Hi, Nothing specific comes to mind.
Did something change either on your PC or on the board? Installed some software or changed SoftConsole project / design etc. ?

kowalski100, 2021-09-03:
No not specifically. It was working at fine at night but suddenly at morning this errors popped up. 
What's you best guess?
Do you think the onboard flashpro6 is faulty? but is detected....

nitindeshpande, 2021-09-03:
Cant guess really! 
As you said, On board eFP6 is definitely detected. That's one less variable :)
We can try the following to narrow it down 
1) Reprogram the job file on the PF SoC device and try debug session.
2) Restart SoftConsole and try debug session again.

ManishaI31214, 2021-12-27:
Try launching a debug session with external FlashPro6 programmer

hughbreslin, 2022-03-03:
Hey, closing this issue as there has been no activity in several months. If this issue persists please add a comment and we can re-open it 🙂

closed 2022-03-03 by hughbreslin