polarfire-soc/polarfire-soc-documentation #12: Kernel panic on mpfs-dev-cli-icicle-kit-es image

NassimEdi, 2022-01-21:
Hello,

I have only added a layer (https://github.com/NobuoTsukamoto/meta-tensorflow-lite) and added some packages to the image. Other than that, the build is still original.
H/W: Icicle Kit rev 1.0 with the 2021.11 reference design.

I get the following kernel log:

```
[5.459010] HSS_Boot_PMPSetupHandler(): Hart1 setup complete

U-Boot 2021.07 (Jul 05 2021 - 15:11:28 +0000)

CPU:   rv64imafdc
Model: Microchip PolarFire-SoC Icicle Kit
DRAM:  1.7 GiB
MMC:   mmc@20008000: 0
Loading Environment from nowhere... OK
In:    serial@20100000
Out:   serial@20100000
Err:   serial@20100000
Net:   eth0: ethernet@20112000
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr.uimg
1263 bytes read in 20 ms (61.5 KiB/s)
## Executing script at 88100000
' - try 'help'd '
6541208 bytes read in 300 ms (20.8 MiB/s)
' - try 'help'd '
## Copying 'fdt-microchip_microchip-mpfs-icicle-kit.dtb' subimage from FIT image at 88300000 ...
sha256+    Loading part 253 ... OK
## Loading kernel from FIT Image at 88300000 ...
   Using 'conf-microchip_microchip-mpfs-icicle-kit.dtb' configuration
   Trying 'kernel-1' kernel subimage
     Description:  Linux kernel
     Type:         Kernel Image
     Compression:  gzip compressed
     Data Start:   0x883000fc
     Data Size:    6525827 Bytes = 6.2 MiB
     Architecture: RISC-V
     OS:           Linux
     Load Address: 0x80200000
     Entry Point:  0x80200000
     Hash algo:    sha256
     Hash value:   a6d72b57aa3205e31251baba09fc131f8783c62e1c1f8e16537639f35b194693
   Verifying Hash Integrity ... sha256+ OK
## Flattened Device Tree blob at 88000000
   Booting using the fdt blob at 0x88000000
   Uncompressing Kernel Image
   Loading Device Tree to 000000008fff9000, end 000000008ffff426 ... OK

Starting kernel ...

[    0.696100] printk: console [ttyS0] enabled
[    0.700452] printk: bootconsole [sbi0] disabled
[    0.706521] 20102000.serial: ttyS1 at MMIO 0x20102000 (irq = 17, base_baud = 9375000) is a 16550A
[    0.717168] 20104000.serial: ttyS2 at MMIO 0x20104000 (irq = 18, base_baud = 9375000) is a 16550A
[    0.727534] 20106000.serial: ttyS3 at MMIO 0x20106000 (irq = 19, base_baud = 9375000) is a 16550A
[    0.737729] [drm] radeon kernel modesetting enabled.
[    0.760666] loop: module loaded
[    0.947954] u-dma-buf udmabuf-ddr-c0: driver version = 3.2.4
[    0.953729] u-dma-buf udmabuf-ddr-c0: major number   = 249
[    0.959298] u-dma-buf udmabuf-ddr-c0: minor number   = 0
[    0.964676] u-dma-buf udmabuf-ddr-c0: phys address   = 0x00000000ae000000
[    0.971531] u-dma-buf udmabuf-ddr-c0: buffer size    = 33554432
[    0.977529] u-dma-buf udmabuf0: driver installed.
[    4.000586] u-dma-buf udmabuf-ddr-nc0: driver version = 3.2.4
[    4.006407] u-dma-buf udmabuf-ddr-nc0: major number   = 249
[... 196 lines cut ...]
[5.508183] RunStateMachine(): boot_service(u54_2) :: SetupPMPComplete -> ZeroInit
[5.517210] RunStateMachine(): boot_service(u54_3) :: SetupPMPComplete -> ZeroInit
[5.526238] RunStateMachine(): boot_service(u54_4) :: SetupPMPComplete -> ZeroInit
[5.535265] RunStateMachine(): boot_service(u54_1) :: ZeroInit -> Download
[5.543598] RunStateMachine(): boot_service(u54_2) :: ZeroInit -> Download
[5.551930] RunStateMachine(): boot_service(u54_3) :: ZeroInit -> Download
[5.560263] RunStateMachine(): boot_service(u54_4) :: ZeroInit -> Download
[5.568596] boot_download_chunks_onEntry(): boot_service(u54_1)::Processing boot image: "u-boot.bin"
[5.578925] RunStateMachine(): boot_service(u54_2) :: Download -> Idle
[5.586911] RunStateMachine(): boot_service(u54_3) :: Download -> Idle
[5.594896] RunStateMachine(): boot_service(u54_4) :: Download -> Idle
[5.646881] RunStateMachine(): boot_service(u54_1) :: Download -> OpenSBIInit
[5.655474] boot_opensbi_init_handler(): boot_service(u54_1)::u54_2:sbi_init 0x80200000
[5.664675] boot_opensbi_init_handler(): boot_service(u54_1)::u54_3:sbi_init 0x80200000
[5.673876] boot_opensbi_init_handler(): boot_service(u54_1)::u54_4:sbi_init 0x80200000
[5.683077] RunStateMachine(): boot_service(u54_1) :: OpenSBIInit -> Wait
[5.691323] boot_opensbi_init_onExit(): boot_service(u54_1)::u54_1:sbi_init 0x80200000
[5.700437] boot_wait_onEntry(): boot_service(u54_1)::Checking for IPI ACKs: - -
[5.709030] boot_wait_handler(): boot_service(u54_1)::Checking for IPI ACKs: ACK/IDLE ACK
[5.718404] RunStateMachine(): boot_service(u54_1) :: Wait -> Idle

```

Thanks,
Nassim

hughbreslin, 2022-02-18:
Hey @NassimEdi! Sorry for the delayed response, are you still having this issue or did you manage to get it resolved? 

Thanks,
Hugh

NassimEdi, 2022-03-21:
This problem was resolved.

closed 2022-03-21 by NassimEdi

Jpgig, 2023-04-06:
@ [NassimEdi] : How did you resolve this issue? I am still facing the same issue?