polarfire-soc/polarfire-soc-documentation #13: Documentation on SBIC?

bkzshabbaz, 2022-02-16:
I'm trying to create a custom MPFSoC design with our own functionality in the FPGA fabric co-existing with the MSS.  Is there documentation on how to generate the Intel-Hex file for programming the eNVM client in Libero?  Is there documentation on the Intel-Hex file header that needs to be pre-pended to the binary ELF file before the MSS will boot from eNVM?  I only found out about this by poking in the bootmode1 directory that gets created in SoftConsole when programming for boot mode 1.  Where does dummySbic.hex come from?  Is there a utility that creates it?

hughbreslin, 2022-02-18:
Hey @bkzshabbaz Have you see the SoftConsole MPFS Boot Mode Programmer documentation? Its available [here](https://mi-v-ecosystem.github.io/SoftConsole-Documentation/SoftConsole-v2021.3/using_softconsole/mpfs.html). The boot mode programmer is bundled with SoftConsole and its used to generate Libero friendly Intel Hex files and you can find the utility and its sources in [SoftConsole Install]/extras/mpfs.

hughbreslin, 2022-02-22:
Excellent I'm glad that helped, were you able to get enough information to close this issue? Would it have helped if our [boot-modes-fundamentals](https://github.com/polarfire-soc/polarfire-soc-documentation/blob/master/fundamentals/boot-modes/boot-modes-fundamentals.md) document mentioned the boot mode programmer in SoftConsole?

bkzshabbaz, 2022-02-23:
Is SoftConsole the only way to get the eNVM programmed with the application?  It definitely would've helped if there was some documentation to describe how Libero can be the last tool used in the development chain.

hughbreslin, 2022-02-24:
You can also program the eNVM using Libero (which is actually what the boot mode programmer in SoftConsole does, using the fpgenprog tool in the Libero installation). To do this you need to open the "Configure Design Initialization Data and Memories" view and in the eNVM tab you can add additional clients to your bitstream:
[image]

To add your application to Libero as a hex file you're probably going to need to run the boot mode programmer in SoftConsole in dry run mode to create a Libero friendly hex file and then add that in. You can find / modify the arguments passed to the tool from the external tool configurations:
[image]

We do have this [document](https://github.com/polarfire-soc/polarfire-soc-documentation/blob/master/software-development/polarfire-soc-software-tool-flow.md#full-flow) with an outline of the design flow but it could probably do with an update to include some more info on the boot modes. Thank you for the feedback, we'll have a look into the docs and how we could make the information clearer :)

bkzshabbaz, 2022-02-24:
> To add your application to Libero as a hex file you're probably going to need to run the boot mode programmer in SoftConsole in dry run mode to create a Libero friendly hex file and then add that in. You can find / modify the arguments passed to the tool from the external tool configurations: 

I think it would be super helpful if what you just said above were placed here: https://github.com/polarfire-soc/polarfire-soc-documentation/blob/master/software-development/polarfire-soc-software-tool-flow.md#programming-envm

I'll leave it up to you to decide where to put it though :)

Thank you.

hughbreslin, 2022-02-28:
Cool thank you for the feedback :)

closed 2022-04-28 by hughbreslin