polarfire-soc/polarfire-soc-documentation #15: HS400 mode on the icicle kit

fcuzzocrea, 2022-03-04:
Hi!

According to ICICLE Kit Documentation (DS60001679B - ICICLE_Kit_User_Guide_VB) the eMMC of the ICICLE Kit (Sandisk SDINBDG4-8G) should support HS400 bus speed mode (coupled with 200 MHz clock speed). However, when setting this either in a baremental application or in the HSS, the MSS_MMC_init function fails to initialize the eMMC. I spotted the issue when trying to configure the eMMC on a custom design and realized that HS400 wasn't working, so I decided to try on the ICICLE Kit and had the same behavior.

Should this be considered as a bug of the BSP or is a not supported mode?

padmaraob, 2022-03-07:
Hi @fcuzzocrea, It supports HS400 mode, will look into the issue and comeback to you

padmaraob, 2022-03-14:
The PolarFire SoC MSS Configurator is setting the eMMC mssio output drive 8mA by default if it change to 10mA then all eMMC bus speed modes work fine including HS400. The Configurator generates XML file that can be used in the project.

Or

Change Bank 4 I/Os value 0x0928 to 0x0930 in the existing XML file like below

MSSIO_BANK4_IO_CFG_0_1_CR:

<register address="0x00000234" description="IO electrical configuration for MSSIO pad" name="MSSIO_BANK4_IO_CFG_0_1_CR">
            <field Type="RW" name="IO_CFG_0" offset="0" width="16">**0x0930**</field>
            <field Type="RW" name="IO_CFG_1" offset="16" width="16">**0x0930**</field>

also change
 
MSSIO_BANK4_IO_CFG_2_3_CR
MSSIO_BANK4_IO_CFG_4_5_CR
MSSIO_BANK4_IO_CFG_6_7_CR
MSSIO_BANK4_IO_CFG_8_9_CR
MSSIO_BANK4_IO_CFG_10_11_CR
MSSIO_BANK4_IO_CFG_12_13_CR

HSS XML : https://github.com/polarfire-soc/hart-software-services/blob/master/boards/mpfs-icicle-kit-es/soc_fpga_design/xml/ICICLE_MSS_mss_cfg.xml

Bare Metal : https://github.com/polarfire-soc/polarfire-soc-bare-metal-examples/blob/main/driver-examples/mss/mss-mmc/mpfs-emmc-sd-write-read/src/boards/icicle-kit-es/fpga_design/design_description/ICICLE_MSS_eMMC_cfg.xml

Once modified XML file then build the example project(HSS or Baremetal)

fcuzzocrea, 2022-03-15:
Hi!!

I just tried what you suggested in the previous comment on my icicle kit, using HSS.

I found out that the HSS half-ignores what is generated with the polarfire soc configurator, as the #define are hardcoded here: https://github.com/polarfire-soc/hart-software-services/blob/01e82e3e05b5b00a22920eba811d95ed885ff16d/services/mmc/mmc_api.c#L66

I changed also the values of those defines to reflect what the configurator tool was setting (0x09300930UL), however I had a very weird behavior.

after HSS boots, my icicle kit first prints this:

`[10.115690] getBootImageFromMMC_(): copyBootImageToDDR_() failed`

then, after some seconds, this is spammed continuously:

`[374.144994] boot_download_chunks_handler(): boot_service(u54_1)::Skipping chunk 0xa01233f8 due to ownership -1`

Do you have any hints to decode this behaviour?

Also, I noticed that on HSS those registers are manually set as part of the initialization stage:

https://github.com/polarfire-soc/hart-software-services/blob/01e82e3e05b5b00a22920eba811d95ed885ff16d/services/mmc/mmc_api.c#L132

I thought that, according to bsp documentation, only the configuration structure should be set https://github.com/polarfire-soc/platform/blob/48c848c6ce788e517b67aab356c1a385799b63a8/drivers/mss/mss_mmc/mss_mmc.h#L372.

So, should I manually set those registers as part of initialization flow?

Thank you!!

padmaraob, 2022-03-29:
Hi @fcuzzocrea,
Can you change eMMC hard coded value **0x09280928UL** to **0x09400940UL** in the mmc_api.c? 
https://github.com/polarfire-soc/hart-software-services/blob/master/services/mmc/mmc_api.c#L66

#define LIBERO_SETTING_MSSIO_BANK4_IO_CFG_0_1_CR_eMMC	0x09400940UL
#define LIBERO_SETTING_MSSIO_BANK4_IO_CFG_2_3_CR_eMMC	0x09400940UL
#define LIBERO_SETTING_MSSIO_BANK4_IO_CFG_4_5_CR_eMMC	0x09400940UL
#define LIBERO_SETTING_MSSIO_BANK4_IO_CFG_6_7_CR_eMMC	0x09400940UL
#define LIBERO_SETTING_MSSIO_BANK4_IO_CFG_8_9_CR_eMMC	0x09400940UL
#define LIBERO_SETTING_MSSIO_BANK4_IO_CFG_10_11_CR_eMMC	0x09400940UL
#define LIBERO_SETTING_MSSIO_BANK4_IO_CFG_12_13_CR_eMMC	0x09400940UL

use 2022.02 release (HSS, icicle-kit-reference-design, meta-polarfire-soc-yocto-bsp)