polarfire-soc/polarfire-soc-documentation #19: ICICLE Kit Stuck

fcuzzocrea, 2022-05-16:
Hi,

Sorry for for the generic title, but this morning, after switching on my icicle kit I found it to be stuck into a weird state. Nothing is output from the UARTs, and the eFP6 is not recognized by my laptop. I tried on different computers, but with the same result. ICICLE Kit yesterday was working completely fine (I was doing some tests with Yocto).

I can't even flash stock design with Libero, as the eFP6 is not recognized. All the KEDs near raspberry pi connector are on, and LPDDR4 led is off, here is a picture of how it looks my board when I switch it on [https://imgur.com/a/WHfmQd8](url).

Is the second time I face this issue (first time I had to buy a new board, but they are not, let's say, cheap).
Do you have any clue how what is happening and why?

Cyril-Jean, 2022-05-16:
I am curious. Do you still have the first faulty Icicle Kit. If you did, could you plug it and send a picture of the powered board?

fcuzzocrea, 2022-05-17:
Yes, I still have it, and behaves the same of this one. Picture of the powered board are available here: https://imgur.com/a/WHfmQd8

fcuzzocrea, 2022-05-18:
Here is another two pictures of both the boards: https://imgur.com/a/yKEGoXE

Cyril-Jean, 2022-05-18:
It looks like 2 power regulators got damaged on both boards. Could you reach out to your local friendly Microchip FPGA field engineer to discuss this?

fcuzzocrea, 2022-05-18:
Thanks, I just contacted him! :)

hughbreslin, 2022-05-20:
Converting this over to a discussion topic :)