polarfire-soc/polarfire-soc-documentation #538: FPGA Configuration and Drivers Problem

BurkayMaras, 2025-04-17:
Hi,  
I'm currently working with the PolarFire Icicle Kit. Previously, the FPGA configuration on my board was modified, but unfortunately I no longer remember what changes were made. I would like to reset or restore the FPGA configuration back to its factory/default state. How can I achieve this?
Additionally, when I connect the board to my PC, it appears under the "Other devices" section in Device Manager as `CP2108 Interface 0, 1, 2, 3` without proper drivers installed.
1. How can I restore the default FPGA configuration on the PolarFire Icicle Kit?
2. What is the correct way to install the CP2108 drivers so that the board is recognized correctly by my system?

- Board: PolarFire SoC Icicle Kit
- Host OS: Windows 10
- Development Tools:SoftConsole 2022.2 
Any documentation, recovery steps, or links related to restoring factory settings and driver installation would be greatly appreciated.  
Thanks in advance!

BurkayMaras, 2025-04-17:
- Update,
I solve driver problem by downloading CP210x Universal Windows Driver from SiliconLab, I run a project with SoftConsole, everything seems normal in SoftConsole terminal but I cannot see any output while I am listening port with putty

rockoguido, 2025-04-17:
> EXTERNAL EMAIL: Do not click links or open attachments unless you know the content is safe

Hi..

You can find pre-built images for the Icicle Kit here..

https://github.com/polarfire-soc/icicle-kit-reference-design/releases

You can use FPExpress (part of Libero) to deploy the image and have a working image in the fabric and the SoC will boot HSS also.

Regards,

Dave Campanella
Principal Embedded Solutions Engineer / FAE
Microchip Technology Inc.
165 Technology Dr
Irvine, CA 92618
+1 925 989 0569 (mobile)
+1 760 304 0172 (office)
***@***.******@***.***>

For any Technical related concerns, kindly you can create web ticket at http://www.microchip.com/support
For Technical Training please go to Microchip University at https://secure.microchip.com/mu

There are several paths to take for Microchip support:
        * Self-paced learning – see the Developer Help Site<http://microchipdeveloper.com/>
        * Knowledge Base - here<http://www.microchip.com/support/hottopics.aspx>
        * Community Forums - help<http://www.microchip.com/forums/> with a lot of information from experts.